syue87/GCodeAnalyser #1: Octoprint plugin

Brandoskey, 2017-12-02:
Any chance of this becoming an octoprint plugin?

syue87, 2017-12-07:
Making it a octoprint plugin is requires a lot of work. The two main problems are:
1. Octoprint plugins are written in Python, the current project is written in Javascript. So I cannot reuse any of the existing code.
2. Octoprint runs on raspberry pi, which has very limited resource comparing to a PC. The code needs to be optimized.
I will consider making it a octoprint plugin when I have some free time, but it will definatly take some time.

joshfria, 2018-01-03:
Why not just have it call on the website with parameters input to a menu?

syue87, 2018-01-04:
@joshfria 
I'm not exactly sure what you mean. If you are talking about redirecting to the website from octoprint plugin, I've though about that. The problem is that the gcode is on the raspberry board, it might take a lot of time to upload the gcode.

joshfria, 2018-01-04:
It shouldn't be any slower to upload than a computer. Gcode is typically under 20-30MB except on really large jobs. Could you not run your javascript through a converter like https://github.com/PiotrDabkowski/Js2Py?

syue87, 2018-01-05:
@joshfria 
The raspberry pi has very limited resource comparing to a PC. Currently the program uses up to a few GB of memory on larger gcodes. In order for the code to run at reasonable speed, I would have to manually write optimized code in python and specifically for the raspberry pi. And like I said, I currently don't have the time to work on this, I will look into it in the future.

derboehsevincent, 2018-04-23:
Just to push this a little bit ;) Python is just the framework in this case. Most of the plugins do the computation stuff in js.

QuickChemist, 2018-06-11:
Hoping for this as well

eyal0, 2018-06-27:
Have you tried running it as-is on RPi 3 with nodejs? What is the speed as compared to desktop?

eyal0, 2018-07-02:
I tested it on RPi.  It's about 8 times slower than my computer.  Not too horrible...

syue87, 2018-07-02:
@eyal0 The program could take more memory then RPi 3 has with large gcode files, in which case the speed will be even slower than that.
The V2 I'm working on will use much less memory, and have APIs to accept data stream from Octoprint plugin.

eyal0, 2018-07-03:
Do you have sample large gcode files that use too much memory? In my testing, all large files worked. I'm also able to run Slic3r on really big files on RPi.

What will the OctoPrint plugin do?

syue87, 2018-07-03:
I don't have the gcode anymore. The one I used for testing is a little over 100MB.
The plug in will add a button beside the gcode files, once clicked it will open a gcodeanalyser window. The code file and settings will be automatically transferred over.

eyal0, 2018-07-04:
Okay, so it would just run the analysis and that's it?  Would it incorporate the results into OctoPrint somehow?  For example, a better estimated print time?

syue87, 2018-07-26:
I can probably write the result back to OctoPrint. I'll have more detail when I start working on it, I haven't really give too much thought yet.

syue87, 2018-11-12:
I'm working on another project that will support gcode processing on the server side, it's a better candidate for Octoprint plugin support. Processing gcode will remain in the browser on the client side for this project.

closed 2018-11-12 by syue87