syue87/GCodeAnalyser #2: Linear advance

Brandoskey, 2017-12-31:
I think there may be an issue with linear advance support? All my gcode I upload with linear advance enabled looks very strange in the layer view. Looks like there's a lot of missing lines or thin lines. Perhaps this is actually intended?

syue87, 2017-12-31:
The program doesn't know whether linear advance is used. It just draws what's represented by the gcodes. 
Linear advance works by start extruding early to build up nozzle pressure, and stop or slow down extruding early to release nozzle pressure. What you are seeing is effect of linear advance so yes it is expected.

Brandoskey, 2018-01-01:
Good to know. Does linear advance affect the time estimation at all? I'm having difficulty getting accurate times on my Prusa MK3. My Maker Select Plus was spot on.

syue87, 2018-01-01:
Linear advance shouldn’t affect time estimation. The first thing I would check is the parameters. Marlin uses mm/min, my program uses mm/sec. Have you divided the marlin value by 60 for the my program?

syue87, 2018-01-23:
@Brandoskey I don't know if you are still having accuracy issueon your estimation, but if you are using M220(speed multiplier) or M221(extrusion multiplier) commands in your gcode, I just fix a problem that could hugely affect estimation accuracy.

closed 2018-01-24 by syue87