syue87/GCodeAnalyser #3: Slic3r Prusa Edition 1.38.7 issue

nickrudolfsky, 2018-01-22:
This gcode analyzer is awesome! Last week, it was estimating the print times on my original Prusa MK3 incredibly accurately (only a couple minutes off for multi-hour prints). But this week I upgraded to the latest version of Slic3r PE (version 1.38.7) and this analyzer stopped working. The rendering is a mess of gigantic blobs, and the print times are crazy high (ex. 3 hours estimated for a print that is less than an hour. 2.5 hours of that 3 hour estimate seems to "retraction time" so that's certainly suspicious).

Prusa must have changed something in the gcode being generated, but I'm not sure what. I know they disabled linear advance in this version, but I don't think that would cause retraction time in this analyzer to spike in such a way. I attached an example gcode file (3D hubs Marvin) that is exhibiting this issue). Any ideas what's going on? If nothing jumps out at you, I might have some time in the future to run your code in a debugger and see where all the time is building up.

Thanks for the great tool!

[3DHubs_Marvin.gcode.zip](https://github.com/syue87/GCodeAnalyser/files/1650805/3DHubs_Marvin.gcode.zip)

syue87, 2018-01-22:
@nickrudolfsky Thanks for submitting this problem, that's a very good issue. I run your gcode though the program and I can see your problem. I'll try to resolve this issue as soon as possible (Hopefully within the week).

nickrudolfsky, 2018-01-22:
Thank you @syue87. That would be great! Let me know if there's anything I can do to help.

syue87, 2018-01-23:
At first glance, I think the program must have parsed some part of the gcode wrong. The reported E axis movement for the attached gcode is 248.21m while the reported Raw Filament Usage is only 0.9m. At where the blobs happen the program must think the it extruded a lot of filament.

syue87, 2018-01-23:
@nickrudolfsky I have fixed your issue. The issue is caused by the "M221 S95" command at the start for setting extrusion multiplier. I forgot to divide the number by 100, so when you use the command the extrusion rate is multiplied by 95. I also made the same mistake on M220 for setting speed multiplier and that's also fixed. I'm surprised I haven't found this problem after months of running this program. Thank you so much for letting me know. 
After the fix your gcode still looks wired on the viewer, and that is due to linear advance. Looks likes it's not disabled in your version.

nickrudolfsky, 2018-01-23:
Ah of course! They did add that M221 command in this new version (to help with over-extrusion issues that users were having). I should have pointed that out. Great job finding the fix and doing it so quickly!

As for the rendering in the viewer, it was showing that behavior before this new version as well. I didn't mention it because I only care about the accurate print time estimate. It doesn't look like linear advance is enabled when the print is running though (the extruder gear doesn't move around as much as it tends to when linear advance is enabled). Whatever!

Thanks again for fixing the issue, and for your great tool!

closed 2018-01-24 by syue87