syue87/GCodeAnalyser #11: [Request] Add ability to export data to excel sheet

morchi83, 2019-07-03:
for example:

columns will be: Layer #, Time to print Layer, Total time from beginning, height, etc...

qalsalaa, 2020-12-03:
Did any Body get it to Export all Tables as Excel? it will be very helpful to share please!
thanks alot

beatwinthewave, 2023-06-20:
Use your browser's dev tools to type "Results[0]" in the javascript console. You'll get the statistics in JSON format, which can then be easily transformed into CSV or EXCEL files. 

```javascript
Results[0]
```

```json
{
    "numberOfLines": 110004,
    "printTime": "2:23:53",
    "accelerationTime": "1:29 (1.0%)",
    "constantSpeedTime": "2:22:24 (99.0%)",
    "moveCommandCount": 77600,
    "reachTargetSpeedCount": "77022 (99.3%)",
    "moveType": "67472 (86.9%) / 6880 (8.9%)",
    "limitedMaxSpeedX": 33404,
    "limitedMaxSpeedY": 25432,
    "limitedMaxSpeedZ": 26,
    "limitedMaxSpeedE": 18123,
    "totalDistance": "119.82 m",
    "totalDistanceX": "75.71 m",
    "totalDistanceY": "69.18 m",
    "totalDistanceZ": "0.05 m",
    "totalDistanceE": "5.46 m",
    "accelerationDistance": "0.95 m (0.8%)",
    "constantSpeedDistance": "118.98 m (99.3%)",
    "printDistance": "97.37 m (81.3%) / 22.45 m (18.7%)",
    "averageSpeed": "13.9 mm/s",
    "averageSpeedX": "8.8 mm/s",
    "averageSpeedY": "8.0 mm/s",
    "averageSpeedZ": "0.0 mm/s",
    "averageSpeedE": "0.6 mm/s",
    "zHopCount": 0,
    "zHopTime": "0:00",
    "retractCount": 3199,
    "retractTime": "35:00",
    "filamentUsage": "3.72 m, 8.96 cm^3",
    "xyFeedrate": "1200 (20.00mm/s) / 48000 (800.00mm/s)",
    "filamentUsageRate": "0.43 mm/s, 2.59 cm/min",
    "filamentLineRatio": "26.14 mm",
    "printSpeed": "18.2 mm/s",
    "travelSpeed": "15.0 mm/s"
}
```