syue87/GCodeAnalyser #13: gcode from  a friend, i see only 2 layer

marcottt, 2020-08-10:
Hi , a friend send me a cimple cube gcode, but it i see on gcodeanalyser, i see only 2 layer...
[40x40x40.zip](https://github.com/syue87/GCodeAnalyser/files/5050793/40x40x40.zip)

attached the file... seem a simple cube....

richfelker, 2020-10-22:
I'm having the same issue with files generated with my current CuraEngine version and configuration, and from reading the source it seems to be that https://github.com/syue87/GCodeAnalyser/blob/49bccddc9d12166f083dee3a649303dcea95b75c/js/gcodeProcessor.js#L582 wrongly classifies layers.

richfelker, 2020-10-22:
I think the problem is that `extrusions` is only pushed to `layer["e"]` when a non-extrusion move is hit, and this misses the case where the layer consists entirely of extrusion moves until the Z move. Normally slicers do some dubious moves before layer change that will make the analyzer happy, and a non-firmware retraction might also make it happy, but firmware-retract or changing layers without a retraction and with no travel moves before the Z move will trigger the bug.

richfelker, 2020-10-22:
OK, @marcottt's testcase is a sort of spiralized print and makes sense that it can't be reasonably split as layers, but it would still make more sense to consider each line as its own layer than to consider the whole spiral part of the print one layer.

The cases where I'm hitting this are not spiralized, just optimized to get rid of bogus travel moves between layers that are causing blobbing.

richfelker, 2020-10-22:
I'll open a separate issue for my problem since the mechanism seems different.

richfelker, 2020-10-22:
Opened related issue #14.

syue87, 2020-10-22:
I am aware of the issue that the viewer does not work with vase mode prints. I've thought about identify layer by Z height, but even if I do that, it still would not work well with vase mode.  Because a curve would be separated in to a lot of layers with very short segments.

closed 2020-10-27 by syue87