syue87/GCodeAnalyser #14: Firmware retraction handling loses layer change

richfelker, 2020-10-22:
When firmware retract is used for the retraction across layer change, the analyzer loses it, combining the whole print into one logical layer. The cause of the problem is:

https://github.com/syue87/GCodeAnalyser/blob/49bccddc9d12166f083dee3a649303dcea95b75c/js/gcodeProcessor.js#L952

where any change to Z that occured while retracted is thrown away and replaced by the old Z at the time of retraction. Instead of `gcode.coord[2] = this.zBeforeFirmwareRetractZhop` it needs to be `gcode.coord[2] -= this.firmwareRetractZhop`. The wrong code just seems to be based on a misunderstanding of `G10`/`G11` semantics; `G11` doesn't save a Z coordinate stored by `G10`; it just reverts the relative change.

richfelker, 2020-10-22:
Indeed that fixes it here.

syue87, 2020-10-22:
Thank you so much for reporting this issue and providing the fix. I will update the website soon.

syue87, 2020-10-22:
Fix is in place. Please let me know if there is any problem. Thank you.

closed 2020-10-22 by syue87

richfelker, 2020-10-22:
It now works. Thanks for the quick fix!