syue87/GCodeAnalyser #15: [Feature Request] Implement support for G2/G3

TodWulff, 2020-11-08:
G2/G3 g-code preprocessing (https://github.com/FormerLurker/ArcWelderLib) is netting a substantive reduction in code size for models with an appreciable amount of arcs/circles.  In looking at these, contrasted with some traditional non-preprocessed G0/G1 encoded models , it seems that GCA doesn't process the G2/G3 constructs properly.?.

https://i.imgur.com/XruIlpb.jpg  G2/G3 on left, G0/G1 on right.

FormerLurker, 2020-12-07:
@syue87, I likely have code that can be pretty easily modified to work with your analyzer.  [Check out this section I added to the Pretty Gcode Viewer](https://github.com/FormerLurker/OctoPrint-PrettyGCode/blob/850bf88d26d456aec303fd280a9356025af4d6d4/octoprint_prettygcode/static/js/prettygcode.js#L1100-L1290).

Eliminateur, 2021-04-30:
+1 from me too